microsoft/devkit-sdk #743: Azure CLI install never worked on macOS

ArthurMa1978, 2018-02-02:
A mac user had to do the manual install process, and the Azure CLI install never worked.  It was missing the packages needed to login, etc.

Last login: Sat Jan 27 10:53:09 on ttys000
Daves-MacBook-Pro:~ dave$ cd ~/Docu
-bash: cd: /Users/dave/Docu: No such file or directory
Daves-MacBook-Pro:~ dave$ cd ~/Documents
Daves-MacBook-Pro:Documents dave$ ls -la
total 64
drwxr-xr-x@  3 dave  staff     96 Jul 19  2016 $RECYCLE.BIN
drwxr-xr-x+ 17 dave  staff    544 Jan 27 09:57 .
drwxr-xr-x+ 58 dave  staff   1856 Jan 27 10:54 ..
-rw-r--r--@  1 dave  staff  26628 Jan 27 09:59 .DS_Store
-rw-------   1 dave  staff      0 Jul  3  2016 .localized
drwxrwxrwx   6 dave  staff    192 Jan 27 09:59 AzureIoTWorkshop
drwxr-xr-x   5 dave  staff    160 Jan 12 07:47 Backgrounds
drwxr-xr-x  71 dave  staff   2272 Dec 20 10:35 Consulting Resources
drwxr-xr-x   9 dave  staff    288 Dec 18 12:39 Continuous Delivery
drwxr-xr-x  13 dave  staff    416 Jan 12 07:14 LeadingAgile
drwxr-xr-x   6 dave  staff    192 Mar 30  2017 Metrics
drwxr-xr-x   7 dave  staff    224 Jan 17  2017 MuseScore2
drwxr-xr-x  13 dave  staff    416 Jan 12 07:14 Personal
drwxr-xr-x  70 dave  staff   2240 Jan 13 20:20 Projects
drwxr-xr-x  20 dave  staff    640 Jan 12 07:11 Software
drwxr-xr-x@  8 dave  staff    256 Oct 24 09:46 Virtual Machines.localized
-rw-r--r--@  1 dave  staff    315 Jan 27 09:57 iot-azure-notes.txt
Daves-MacBook-Pro:Documents dave$ cd AzureIoTWorkshop/
Daves-MacBook-Pro:AzureIoTWorkshop dave$ cd DevKit
Daves-MacBook-Pro:DevKit dave$ ls -la
total 24
drwxrwxrwx  5 dave  staff   160 Jan 27 09:59 .
drwxrwxrwx  6 dave  staff   192 Jan 27 09:59 ..
-rw-r--r--@ 1 dave  staff  8196 Jan 27 09:59 .DS_Store
drwxrwxrwx  8 dave  staff   256 Jan 27 09:28 devkit_install_mac_1.3.1.42
drwxrwxrwx  8 dave  staff   256 Jan 27 09:28 devkit_install_win_1.3.1.42
Daves-MacBook-Pro:DevKit dave$ cd devkit_install_mac_1.3.1.42/
Daves-MacBook-Pro:devkit_install_mac_1.3.1.42 dave$ brew install arduino
Error: No available formula with the name "arduino" 
==> Searching for a previously deleted formula (in the last month)...
Warning: homebrew/core is shallow clone. To get complete history run:
  git -C "$(brew --repo homebrew/core)" fetch --unshallow

Error: No previously deleted formula found.
==> Searching for similarly named formulae...
==> Searching local taps...
Error: No similarly named formulae found.
==> Searching taps...
==> Searching taps on GitHub...
Error: No formulae found in taps.
Daves-MacBook-Pro:devkit_install_mac_1.3.1.42 dave$ brew cask reinstall --force arduino
==> Creating Caskroom at /usr/local/Caskroom
==> We'll set permissions properly so we won't need sudo in the future
Password:
==> Caveats
arduino requires Java. You can install the latest version with

  brew cask install java

==> Satisfying dependencies
==> Downloading https://downloads.arduino.cc/arduino-1.8.5-macosx.zip
######################################################################## 100.0%
==> Verifying checksum for Cask arduino
==> Installing Cask arduino
==> Moving App 'Arduino.app' to '/Applications/Arduino.app'.
==> Linking Binary 'arduino-builder' to '/usr/local/bin/arduino-builder'.
馃嵑  arduino was successfully installed!
Daves-MacBook-Pro:devkit_install_mac_1.3.1.42 dave$ ./install.sh

Updaing brew package list...

==> Reinstalling yarn 
[... 321 lines cut ...]
                       snapshots.
    sql              : Manage Azure SQL Databases and Data Warehouses.
    storage          : Manage Azure Cloud Storage resources.
    tag              : Manage resource tags.
    vm               : Provision Linux or Windows virtual machines.
    vmss             : Manage groupings of virtual machines in an Azure Virtual Machine Scale Set
                       (VMSS).
    webapp           : Manage web apps.
Daves-MacBook-Pro:devkit_install_mac_1.3.1.42 dave$ az --version
No module named 'pkg_resources'
Traceback (most recent call last):
  File "/usr/local/Cellar/azure-cli/2.0.25/libexec/lib/python3.6/site-packages/knack/cli.py", line 187, in invoke
    self.show_version()
  File "/usr/local/Cellar/azure-cli/2.0.25/libexec/lib/python3.6/site-packages/azure/cli/core/__init__.py", line 79, in show_version
    print(get_az_version_string())
  File "/usr/local/Cellar/azure-cli/2.0.25/libexec/lib/python3.6/site-packages/azure/cli/core/util.py", line 60, in get_az_version_string
    installed_dists = get_installed_cli_distributions()
  File "/usr/local/Cellar/azure-cli/2.0.25/libexec/lib/python3.6/site-packages/azure/cli/core/util.py", line 51, in get_installed_cli_distributions
    from pkg_resources import working_set
ModuleNotFoundError: No module named 'pkg_resources'
Daves-MacBook-Pro:devkit_install_mac_1.3.1.42 dave$ azure-cli(2.0.20)
-bash: syntax error near unexpected token `2.0.20'
Daves-MacBook-Pro:devkit_install_mac_1.3.1.42 dave$ az login
Error loading command module 'dla'
To sign in, use a web browser to open the page https://aka.ms/devicelogin and enter the code F4HS47CN2 to authenticate.

Sneezry, 2018-04-13:
Related issue on Azure CLI: https://github.com/Azure/azure-cli/issues/5503

ihere1, 2018-04-24:
Close issue because it is a user environment problem. See how to fix at previous comment.

closed 2018-04-24 by ihere1