microsoft/devkit-sdk #744: Windows user that could not get Visual Studio code to recognize the MXChip/port

ArthurMa1978, 2018-02-02:
3I had a Windows user that could not get Visual Studio code to recognize the MXChip/port.  We tried un-installing, re-installing, re-selecting the board manager both in VS Code as well as in the Arduino IDE, no luck.   In VS Code, when trying to select the MXChip board, the Upload Method selection wasn’t visible:

[image]

lirenhe, 2018-02-07:
I would suggest updating the document on how to workaround this issue.
Just one line change in the config file.
I worked with people for Arduino extension but still unable to figure out the root cause.

BretStateham, 2018-02-07:
What is the "one line" to change in the config file?

lirenhe, 2018-02-08:
@BretStateham , open arduino.json under .vscode folder for the mini solution, modify the configuration part to the following line:
    "configuration": "upload_method=OpenOCDMethod"

ArthurMa1978, 2018-02-09:
Will provide a FAQ

lirenhe, 2018-03-07:
FAQ added in https://microsoft.github.io/azure-iot-developer-kit/docs/faq/

closed 2018-03-07 by lirenhe