microsoft/devkit-sdk #745: The node.js not in the system PATH env

ArthurMa1978, 2018-02-02:
I had another attendee where they installed node.js via the devkit, however it would never resolve inside the VS Code integrated terminal (even after closing and restarting VS code).  I had her manually install node from the web site, making sure to include Node in the path, and that fixed it.

ArthurMa1978, 2018-02-09:
Try to repo this issue first

v-fional, 2018-02-26:
We tried the following scenarios, but hard to repo this issue.

1. If uninstall nodejs, and make sure the variable is removed from the Path env, then run the installation script, the nodejs is installed with variable added to the Path env

2. Just remove the variable from the Path env, then run the installation script, the installation is failed with "Error occurred when running yarn install" error.

ArthurMa1978, 2018-04-08:
Rare case and hard to repo, and since we are working on the new VS Code extension to replace the installation script, so close this issue.

closed 2018-04-08 by ArthurMa1978