microsoft/devkit-sdk #752: The path of error file incorrect

v-fional, 2018-02-07:
Package version: 1.3.2.44
Platform: Windows and Mac

when running examples, (such as getstarted), the following error message displays.

----------Error Message-----------------

201.\features\mbedtls\src\ssl_tls.c (3789): mbedtls_ssl_fetch_input() returned -27648 (-0x6c00)

.\features\mbedtls\src\ssl_tls.c (3734): mbedtls_ssl_read_record_layer() returned -27648 (-0x6c00)

.\features\mbedtls\src\ssl_tls.c (6577): mbedtls_ssl_read_record() returned -27648 (-0x6c00)

8-02-07 02:55:12 ERROR: File:C:\Users\v-fional\AppData\Local\Arduino15\packages\AZ3166\hardware\stm32f4\1.3.0\cores\arduino\az_iot\c-utility\adapters\socketio_mbed_os5.c Func:retrieve_data Line:172, Socketio_Failure: underlying IO error -3004.
2018-02-07 02:55:13 INFO:  >>>Re-connect.
2018-02-07 02:55:13 ERROR: File:C:\Users\v-fional\AppData\Local\Arduino15\packages\AZ3166\hardware\stm32f4\1.3.0\cores\arduino\az_iot\umqtt\src\mqtt_client.c Func:sendPacketItem Line:332, 620: Failure sending control packet data
2018-02-07 02:55:13 ERROR: File:C:\Users\v-fional\AppData\Local\Arduino15\packages\AZ3166\hardware\stm32f4\1.3.0\cores\arduino\az_iot\umqtt\src\mqtt_client.c Func:mqtt_client_disconnect Line:1176, Error: mqtt_client_disconnect send failed

ArthurMa1978, 2018-03-07:
the PR is here #753

v-fional, 2018-03-12:
It is hard to repro this issue now.

v-fional, 2018-03-14:
Tested on the latest code, the issue is fixed, only the file name displays now,

closed 2018-04-08 by ArthurMa1978