microsoft/devkit-sdk #757: fatal error: Riot.h: No such file or directory  #include "Riot.h"

adumont, 2018-02-12:
Verifying...
In file included from /home/adumont/.arduino15/packages/AZ3166/hardware/stm32f4/1.3.2/libraries/AzureIoT/src/DevkitDPSClient.cpp:6:0:
/home/adumont/.arduino15/packages/AZ3166/hardware/stm32f4/1.3.2/cores/arduino/az_iot/provisioning_client/deps/RIoT/Reference/RIoT/Core/RiotCore.h:14:18: fatal error: Riot.h: No such file or directory
 #include "Riot.h"
                  ^
compilation terminated.
exit status 1
[Error] Exit with code=1

adumont, 2018-02-12:
Fixed in https://github.com/Microsoft/devkit-sdk/pull/756

ArthurMa1978, 2018-03-07:
Thanks  Alexandre, this will be in [1.3.3 release](https://github.com/Microsoft/devkit-sdk/issues/762).

v-fional, 2018-03-12:
Tested on the latest code, it works well without any exception

adumont, 2018-04-04:
Fixed in #798

closed 2018-04-08 by ArthurMa1978