microsoft/devkit-sdk #796: Do not send project name in plain text

chyuanmsft, 2018-03-29:
The tasks now send project name in telemetry to help us understand usage of our mini solutions. After discussed offline, please hash the project name before sending telemetry.

chyuanmsft, 2018-03-30:
Please use a new property "hash_project_name" to save hashed value so we don't need to worry about what the project_name column stores.

czgtest, 2018-04-19:
[image]

[image]

[image]

czgtest, 2018-04-20:
Fix work now 
[image]

closed 2018-04-20 by czgtest