microsoft/devkit-sdk #802: Update IoT Hub installer such that it works for users who only have resource group-level access

KirkMunro, 2018-04-09:
The IoT development kit includes tasks that are added to Visual Studio Code that will deploy an IoT Hub into Azure. If the user you use to connect into Azure while following these tasks only has resource group-level access (i.e. no subscription access), the deployment will fail unnecessarily because it will attempt to register the Microsoft.Devices resource provider even if it is already registered in the subscription, in the context of a user who only has resource group-level access.

The root issue lies on line 21 of task-iothub.js. Instead of lazily assuming you are a subscription admin and forcing the registration of the Microsoft.Devices resource provider, the logic should check to see if the Microsoft.Devices resource provider is already registered or not before attempting to register the resource provider. This approach allows a subscription admin to register the Microsoft.Devices resource provider so that other users who do not have that level of access can successfully deploy IoT Hubs into the resource groups where they have access.

To fix this, replace this line (line 21):

```javascript
_context.next = 7;return azurecli.exec('provider register -n "Microsoft.Devices"');case 7:
```

With the following lines:

```javascript
_context.next = 5;return (azurecli.execResultJson('provider show -n "Microsoft.Devices"')); case 5:
devicesProvider = _context.sent;_context.next = 7;if (devicesProvider.registrationState === 'NotRegistered') {
    console.log('Registering resource provider "Microsoft.Devices" in subscription ' + context.subscription);
    return azurecli.exec('provider register -n "Microsoft.Devices"')};break;case 7:
```

I would have made these changes in a PR myself, but @ArthurMa1978 informed me on Gitter that the task part of the device SDK is not open source at the moment and requested I log the issue here so that it can be addressed in the new VS Code extension that is coming soon. I already made these changes locally and it seems to work just fine using a user who only has contributor access to a single resource group in Azure, with no subscription-level permissions at all.

ArthurMa1978, 2018-04-16:
Thanks @KirkMunro for your contribution, we have merged your fix, and will be released in next version.

v-fional, 2018-04-16:
Tested with the latest task code, the user can cloud-provision successfully if only has resource group access

KirkMunro, 2018-04-16:
@ArthurMa1978 @v-fional Thank you both, I really appreciate you accepting this improvement!

ArthurMa1978, 2018-04-17:
will ship it in 1.3.5

closed 2018-04-17 by ArthurMa1978