microsoft/devkit-sdk #821: memory usage API doesn't work

obastemur, 2018-05-04:
Example below prints `0` for all;

```
        mbed_stats_heap_t heap;
        mbed_stats_stack_t stack;
        mbed_stats_heap_get(&heap);
        mbed_stats_stack_get(&stack);
        char bufff[STRING_BUFFER_32] = {0};
        sprintf(bufff, "cs %d ac %d", heap.current_size, heap.alloc_cnt);
        Screen.print(0, "heap");
        Screen.print(1, bufff);
        Screen.print(2, "stack");
        memset(bufff, 0, STRING_BUFFER_32);
        sprintf(bufff, "rs %d sc %d", stack.reserved_size, stack.stack_cnt);
        Screen.print(3, bufff);
```

v-fional, 2018-05-11:
The related file is updated, it can work correctly now, please refer to PR https://github.com/Microsoft/devkit-sdk/pull/825

v-fional, 2018-05-25:
The test case is added, please refer to RP [#836](https://github.com/Microsoft/devkit-sdk/pull/836)

ArthurMa1978, 2018-06-11:
The heap status API is ok, but stack has problem which is a underlying mbedOS issue, need to upgrade the mbedOS to 5.7, open another issue #849 to track it.

closed 2018-06-11 by ArthurMa1978