microsoft/devkit-sdk #822: Issue with HttpClient and Audio

lgermain91, 2018-05-08:
I try to play wav files from an url.

It works first time for small file, but fails then.

Also noticed that when loading files > 20k  then it doesnt work

Any adivce?

HTTPClient client = HTTPClient(HTTP_GET, "http://www.pacdv.com/sounds/people_sound_effects/yes_1.wav");
  //http://www.pacdv.com/sounds/domestic_sound_effects/microwave-oven-timer-1.wav
  //http://www.pacdv.com/sounds/people_sound_effects/yes_1.wav
 //HTTPClient client = HTTPClient(HTTP_GET, "http://www.wavsource.com/snds_2018-01-14_3453803176249356/animals/pig4.wav");
 Screen.print(0, "Loading Audio File");
   response = client.send();
   Screen.print(0, "Loaded");
//delay(2000);

char buffer [500];
 if (response != NULL && response->status_code == 200)
{
      int vn =sprintf (buffer, "R:%i s:%i ",response->status_code,response->body_length);
      Screen.print( 2,buffer,true);
 Screen.print( 0,"Audio Loaded OK",true);
 strcpy(duplicate,(char*)response->body);
  for ( int k=0;k<3;k++)
 {
  Audio.startPlay ((char*)response->body, response->body_length-1);
  
   int c=0;
   while (Audio.getAudioState() == AUDIO_STATE_PLAYING)
  {
    c++;
    int vn =sprintf (buffer, "cpt :%i %i",c,Audio.getCurrentSize());
    Screen.print( 0,buffer); 
  }

ihere1, 2018-05-09:
Caused by can not malloc enough space. Make sure you can malloc at least two times memory of the http-response before you call HTTP_GET

lgermain91, 2018-05-09:
thx,

Do you have a snippet example on how to do a malloc on the repose or response-> body ?

Thx

lgermain91, 2018-05-09:
tried this
 response = (Http_Response*)malloc(500000);

but  i think the malloc should be on response->body

ihere1, 2018-05-11:
The malloc was on response->body, called when each frame of message received. Malloc in ino and then release (as I mentioned in last comment) is only a test to see weather there is enough space for response. Each time a frame of message received, we will malloc a new buffer, copy all data to new buffer, and then free old buffer. So two times of space is needed.
Another way is that you provide a callback function on HTTPClient created, and process received message in callback. By this way, there will be no response -> body.

closed 2018-05-21 by ihere1