microsoft/devkit-sdk #823: Security Chip

abhatikar, 2018-05-09:
Hi,

Is there a sample showing how to use the security chip for the Az3166 board ?
A small sample code would be very helpful.
thanks in advance,

ihere1, 2018-05-11:
API reference at https://microsoft.github.io/azure-iot-developer-kit/docs/apis/EEPROMInterface/

ihere1, 2018-05-14:
The secure channel can not be disabled. It is a by design feature for secure. And the behavior before secure channel enabled and after secure channel enabled is the same. The only difference is data storage format on secure chip and the application code can not tell whether secure channel is enabled. Secure channel is to make sure host board(devkit) and secure chip can verify each other in industrial environment. 
You can enable secure channel any time. Do you need a sample code with only read and write ?

abhatikar, 2018-05-14:
Thanks for your reply @ihere1.Sure if you could share some sample that would be helpful. I did not quite get the idea and benefit of having the secured chip.

ihere1, 2018-05-15:
After secure channel enabled, the data between secure chip and board will be encrypted. Other board without specific key can not read data from secure chip. But it is hard to demo. For devkit, the secure chip is still a storage even after secure channel enabled. Encryption and decryption were completed in lower level library.

closed 2018-05-21 by ihere1