microsoft/devkit-sdk #835: The rule of temperatureAlert is wrong

ArthurMa1978, 2018-05-24:
Should be like this:

```DevKitMQTTClient_Event_AddProp(message, "temperatureAlert", temperatureAlert > 30 ? "true" : "false");```

v-fional, 2018-05-25:
code is updated, please refer to PR #[837 ](https://github.com/Microsoft/devkit-sdk/pull/837)

v-fional, 2018-06-08:
Checked on version 1.3.7, it is fixed so close it

ihere1, 2018-06-11:
Following device message shows that the original code is correct. Please help revert the code.

6/11/2018 10:56:23 AM> Device: [AZ3166], Data:[{
    "deviceId": "AZ3166",
    "messageId": 20,
    "temperature": 31.299999,
    "humidity": 75.099998
}]Properties:
'temperatureAlert': 'true'

6/11/2018 10:56:34 AM> Device: [AZ3166], Data:[{
    "deviceId": "AZ3166",
    "messageId": 21,
    "temperature": 30.100000,
    "humidity": 65.400002
}]Properties:
'temperatureAlert': 'true'

6/11/2018 10:56:45 AM> Device: [AZ3166], Data:[{
    "deviceId": "AZ3166",
    "messageId": 22,
    "temperature": 29.400000,
    "humidity": 63.700001
}]Properties:
'temperatureAlert': 'false'

ihere1, 2018-06-11:
https://github.com/Microsoft/devkit-sdk/pull/851

closed 2018-07-19 by ArthurMa1978