microsoft/devkit-sdk #863: useless comparisons

Adamantinu, 2018-07-05:
In the GetStarted Example there are useless comparisons like
 if (humidity != humidity) 
{
json_object_set_null(root_object, "humidity");
}

Those could be confusing for starters with the DevKit.
Please remove them.

ArthurMa1978, 2018-07-11:
not useless, send the humidity to IoT hub only it has changed

Adamantinu, 2018-07-11:
float humidity = readHumidity(); //Here you read the value of the humidity sensor
...
if (humidity != humidity) //Here you compare the value of the humidity Sensor you read out, to its self
{} //Therfore this Code will never be executed since the same variable can't have two values at once

Please correct me if I'm wrong, since i am normally developing in C#

ArthurMa1978, 2018-07-17:
My bad, didn't review the code carefully.
Stupid bug!!! will fix it soon.

Adamantinu, 2018-07-17:
No Problem, I'm happy to help.

closed 2018-08-17 by ArthurMa1978