microsoft/devkit-sdk #971: HttpClient not working with HTTPS urls

DerekFoster1976, 2019-02-24:
Cannot get any HTTPS REest calls to work. Have tried multiple APIs with correct cert fingerprint. Even the official sample code for HttpClient using HTTPS [here](https://microsoft.github.io/azure-iot-developer-kit/docs/apis/http-client/) doesn't work either. Standard HTTP calls work fine. There is a lot of discussion around this for standard Arduino sketches with mixed results. A solution would be great as essentially can't call most APIs directly from the board as they all use HTTPS.

ArthurMa1978, 2019-03-07:
@DerekFoster1976  thank you for feedback.
Yes, this HttpClient doesn't support HTTPS, add this to feature backlog

JerryYangKai, 2019-12-03:
Hi @DerekFoster1976 , We have released DevKit SDK 1.6.5 and HttpClient supporting HTTPS now. Please try it.

closed 2019-12-03 by JerryYangKai