microsoft/devkit-sdk #1009: MXCHIP AZ3166 serial port issue

br1pro, 2019-07-04:
I'm following this tutorial https://microsoft.github.io/azure-iot-developer-kit/docs/get-started/ to connect my AZ3166 to Iot Hub. 
The tutorial explain how to run the "Get started" tutorial.
In the section "Build your first project" I'm blocked at point 5. If I click on the "select serial port" button in the right down corner, nothing happens.
I'm using the last firmware 1.6.2 and visual studio code on a win10 pc.

br1pro, 2019-07-05:
[image]

br1pro, 2019-07-05:
The serial port works well if I try to open it from Putty, so I think this is a VScode issue

ArthurMa1978, 2019-07-10:
https://github.com/microsoft/vscode-iot-workbench/issues/661

fixed since the VS Code workbench v0.3.1

closed 2019-07-10 by ArthurMa1978