microsoft/devkit-sdk #1022: Shutdown command invoked

baqwas, 2019-08-12:
Hello,
I using the example _ScanNetworks_ without modifications. The code compiles, the code upload succeeds but the output window has the following lines in tail section:
```
Sketch uses 195796 bytes (18%) of program storage space. Maximum is 1048576 bytes.
Global variables use 53388 bytes (20%) of dynamic memory, leaving 208756 bytes for local variables. Maximum is 262144 bytes.
GNU ARM Eclipse 64-bits Open On-Chip Debugger 0.10.0-00113-g0f83948 (2017-01-24-19:18)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
hla_swd
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
adapter speed: 2000 kHz
adapter_nsrst_delay: 100
none separate
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Info : clock speed 1800 kHz
Info : STLINK v2 JTAG v28 API v2 SWIM v17 VID 0x0483 PID 0x374B
Info : using stlink api v2
Info : Target voltage: 3.269076
Info : stm32f4x.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
adapter speed: 1800 kHz
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x08000034 msp: 0x200073fc
Info : Unable to match requested speed 8000 kHz, using 4000 kHz
Info : Unable to match requested speed 8000 kHz, using 4000 kHz
adapter speed: 4000 kHz
** Programming Started **
auto erase enabled
Info : device id = 0x30006441
Info : flash size = 1024kbytes
target halted due to breakpoint, current mode: Thread 
xPSR: 0x61000000 pc: 0x20000046 msp: 0x200073fc
wrote 212992 bytes from file /tmp/arduino_build_607412/ScanNetworks.ino.bin in 4.969891s (41.852 KiB/s)
** Programming Finished **
** Verify Started **
target halted due to breakpoint, current mode: Thread 
xPSR: 0x61000000 pc: 0x2000002e msp: 0x200073fc
verified 197380 bytes in 0.377446s (510.679 KiB/s)
** Verified OK **
** Resetting Target **
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
adapter speed: 1800 kHz
shutdown command invoked

```
Only the power (green) led remains lit. Subsequently, I can run _Blink_ without issue but reverting back to _ScanNetworks_ (or any other example other than _Blink_) seems to initiate **shutdown** of all things. What is the workaround, please? Thanks.

Kind regards.

dooriya, 2019-09-26:
@baqwas Thank you for raising this issue. We are currently investigating and will update you shortly.

JerryYangKai, 2019-09-29:
@baqwas Hi! I am trying to repro your issue but I haven't seen shutdown in my devkit. So can you show me more information about your project, like your settings in c_cpp_properties.json?

the following lines are part of my output when upload
 `Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
adapter speed: 2000 kHz
adapter_nsrst_delay: 100
none separate
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Info : clock speed 1800 kHz
Info : STLINK v2 JTAG v28 API v2 SWIM v17 VID 0x0483 PID 0x374B
Info : using stlink api v2
Info : Target voltage: 3.293927
Info : stm32f4x.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
adapter speed: 1800 kHz
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x08000034 msp: 0x2000d650
Info : Unable to match requested speed 8000 kHz, using 4000 kHz
Info : Unable to match requested speed 8000 kHz, using 4000 kHz
adapter speed: 4000 kHz
** Programming Started **
auto erase enabled
Info : device id = 0x30006441
Info : flash size = 1024kbytes
target halted due to breakpoint, current mode: Thread 
xPSR: 0x61000000 pc: 0x20000046 msp: 0x2000d650
wrote 344064 bytes from file C:\Users\kaiyan\AppData\Local\Temp\arduino_build_895940/ScanNetworks.ino.bin in 8.087001s (41.548 KiB/s)
** Programming Finished **
** Verify Started **
target halted due to breakpoint, current mode: Thread 
[Done] Uploaded the sketch: ScanNetworks.ino`

and in Serial Monitor
`Scanning available networks...
** Scan Networks **
number of available networks:10
0) Ruff_R0101965	Signal: -60 dBm	Encryption: WPA2
1) MSFTINET	Signal: -68 dBm	Encryption: WPA2
2) MSFTCORP	Signal: -68 dBm	Encryption: WPA2
3) MSFTGUEST	Signal: -69 dBm	Encryption: None
4) MSFTCORP	Signal: -69 dBm	Encryption: WPA2
5) MSFTINET	Signal: -69 dBm	Encryption: WPA2
6) laptop_jiaqi	Signal: -70 dBm	Encryption: WPA2
7) MSFTGUEST	Signal: -70 dBm	Encryption: None
8) MSFTINET	Signal: -77 dBm	Encryption: WPA2
9) MSFTGUEST	Signal: -77 dBm	Encryption: None`

closed 2019-10-12 by JerryYangKai