microsoft/devkit-sdk #1023: WebServer available method blocks until a client connects

fmuntean, 2019-08-16:
Trying the WiFiWebServer sample and in the loop method when calling 
WiFiClient client = server.available();
The thread is blocked here until the client connects to the server. This makes impossible to run anything else on the board.

Please make this method return null pointer when no client is connected to the server.

dooriya, 2019-09-26:
@fmuntean Thanks for the feedback!

Currently, the underlining `WiFiServer.available()` is implemented as a **BLOCKING** operation in the Devkit SDK, which means the server would block until incoming client connection occurs.

Behind the scene, we leverage the [TCPServer from mbed-os](https://os.mbed.com/docs/mbed-os/v5.9/mbed-os-api-doxy/class_t_c_p_server.html) to implement the core function of the WifiServer class, which also use a blocking mode by default.

The improvement here can be providing the option to `set_timeou`t or `set_blocking` for user. Then user can choose whether to use a blocking TCPServer or not according to his specific scenario.

JerryYangKai, 2019-12-03:
Hi @fmuntean,   We have released DevKit SDK 1.6.5 and added  `set_timeout` option for server.available(). Please try it.

closed 2019-12-03 by JerryYangKai