microsoft/devkit-sdk #1029: [Feature Request] Ability to join enterprise WiFI

sfletc, 2019-09-07:
Unless I'm missing something, this library doesn't allow for joining an enterprise WiFi network such as eduroam, where a username and password are required.  

Would be nice feature, as the device could then be used at universities as a great IOT learning platform.

liydu, 2019-09-09:
@sfletc , DevKit supports connecting WiFi via username / password: https://docs.microsoft.com/en-us/azure/iot-hub/iot-hub-arduino-iot-devkit-az3166-get-started. Are you talking about the captive portal like those in Starbucks which you connect to a hotspot and it pops up a page for username and password?

sfletc, 2019-09-11:
WPA2 Enterprise Wifi requires an SSID, username and password.  It's not the same as in the getting started guide (which is fine for a home WPA2 router) or a captive portal like starbucks.  

Authentication is via EAP-TTLS or PEAP on eduroam, for example.  Other devices like the ESP32 are capable (e.g. https://github.com/espressif/esp-idf/tree/master/examples/wifi/wpa2_enterprise); just trying to work out whether this one is, so it can be used as a learning tool.

liydu, 2019-09-17:
DevKit uses cypress bcm43362, unfortunately, it does not support WPA2 Enterprise.

dooriya, 2019-09-25:
Close this issue since it is out-of-scope

closed 2019-09-25 by dooriya