microsoft/devkit-sdk #1053: millis() value less then 4294967 (about 1.2hours)

matsujirushi, 2019-11-22:
Return type of millis() is uint32_t.  
0~2^32-1.

But I got maximum value of millis() is (2^32-1)/1000.

Please see this [code](https://github.com/microsoft/devkit-sdk/blob/master/AZ3166/src/cores/arduino/wiring.cpp#L26).

JerryYangKai, 2019-11-25:
@matsujirushi  I have checked the code, and found us_ticker_read() return type is uint32_t, so its maximum value is 2^32-1. And in function millis(), it divide 1000 of us_ticker_read()'s return , so millis() can only return (2^32-1)/1000.

matsujirushi, 2019-11-26:
I see.
I want to fix it because Arduino API defined [that](https://www.arduino.cc/reference/en/language/functions/time/millis/).

JerryYangKai, 2019-11-26:
@matsujirushi  Sure! I suggest rename current millis() to anothor name and you can write a new millis() as Arduino API defined. Thanks!

JerryYangKai, 2019-12-09:
@matsujirushi  Since 1.6.5 is a stable version, so I will fixed it in the next released. You can change your local sdk for temporary work around.

JerryYangKai, 2019-12-11:
Hi @matsujirushi, I checked the open source mbed code and found there are only 32 bit to store the counter, so it has limited in millis(). We have already provided an api for millis count, it is in [SystemTickCounter.h](https://github.com/microsoft/devkit-sdk/blob/master/AZ3166/src/cores/arduino/system/SystemTickCounter.h).

closed 2019-12-16 by JerryYangKai