microsoft/devkit-sdk #1061: Can't connect to the IoT Hub after config with set_az_iothub

lokijota, 2020-02-09:
Just the kit a few days ago and have been playing with it. After reflashing, I have it connected to wifi and the screen gives me the different readings.

I logged into the serial interface and ran version:

```
DevKitSDK version: 1.6.5
Mico version: 31620002.049
mbed-os version: 5.4.3
mbed TLS version: 2.15.0
WIFI version: wl0: Sep 10 2014 11:28:46 version 5.90.230.10
```
Because the UI configuration page on the device doesn't ask me for the device connection, I ran set_az_iothub over the serial interface:

```
# set_az_iothub "HostName=xxxxx.azure-devices.net;DeviceId=mx1;SharedAccessKey=xxxx"
INFO: Set Azure Iot hub connection string successfully.
```

But when the device reboots I always get:

`Failed to send telemetry data: 400` in the serial, the blue light doesn't turn on, and if I monitor the messages with the `az iot hub monitor-events` I don't get anything from the devkit (only from other devices).

I tried running `set_az_iothub` passing the connection string without the commas and had the same result.

Other devices I have in my IoT Hub work fine and send telemetry. I am SURE the device connection string is correct, by the way: I created a device, went into the page, and copied the primary.

I'd appreciate any help!

JerryYangKai, 2020-02-12:
Hi @lokijota , I need more information to resolve this issue, you can check if this connection string is correct frist. I suggest use workbench command in vscode to config this connection string for work around.

lokijota, 2020-02-13:
Hey Jerry,

here's how I got the device connection string:

1- Opened portal.azure.com
2- Accessed my IoT Hub
3- Clicked "IoT Devices" in menu section "Explorers"
4- Clicked the device I created (I call it "mx1") in the list of devices. The status is Enabled, Authentication Type = SAS, and columns "Last status update" and "Cloud do Device Message Count" are empty
5- In the detail page of the device, "Enable connection to IoT Hub" = Enabled.
6- Clicked the "Copy" icon to the right of the "Primary Connection String" field: `HostName=myiothub.azure-devices.net;DeviceId=mx1;SharedAccessKey=mykey`

Regarding using the Workbench, I struggled with understanding how it works. I have it installed in VS Code but when I launched the command to "Configure device settings", I ended up having to create a project which generated an .ino file. Where do I test the connection string? 

I then followed the instructions here: https://docs.microsoft.com/en-us/azure/iot-hub/tutorial-connectivity and managed to succesfully test connectivity:

```
node SimulatedDevice-1.js "<the key I got from above>"
IoT Hub troubleshooting tutorial
Simulated device #1

Client connected
```

What other information can I provide?

PS: While trying to understand how to use the Workbench I noticed:

- in https://marketplace.visualstudio.com/items?itemName=vsciot-vscode.azure-iot-tools, the link to "get started" is broken;
- in https://github.com/microsoft/vscode-iot-workbench the link to "Use Azure IoT Device Workbench extension in Visual Studio Code" is also broken.

Where can one find documentation on how to use the IoT Workbench in VS Code?

JerryYangKai, 2020-02-13:
@lokijota  Hi, thanks for your suggestions. Let us come to the beginning.  Have you tried this format?
```
# set_az_iothub HostName=xxxxx.azure-devices.net;DeviceId=mx1;SharedAccessKey=xxxx
```
I mean remove these quotes and check if it works.

For the second questions. You can follow this tutorials to try our workbench
https://docs.microsoft.com/en-us/samples/azure-samples/mxchip-iot-devkit-get-started/sample/

JerryYangKai, 2020-02-19:
Since this issue has no reply during a week. I close this issue.

lokijota, 2020-02-22:
Hi Jerry. As per my initial message: _"I tried running set_az_iothub passing the connection string without the commas and had the same result"._

Issue is NOT closed, and there's people in other threads mentioning the same issue. Please re-open.

JerryYangKai, 2020-02-24:
@lokijota  Sorry about that, I have reopened this issue. 
Fristly, I ran set_az_iothub over the serial interface in my mxchip (sdk version 1.6.5). This is my output log
`
************************************************
** MXChip - Microsoft IoT Developer Kit **
************************************************
Configuration console:
 - help: Help document.
 - version: System version.
 - exit: Exit and reboot.
 - scan: Scan Wi-Fi AP.
 - set_wifissid: Set Wi-Fi SSID.
 - set_wifipwd: Set Wi-Fi password.
 - set_az_iothub: Set IoT Hub device connection string.
 - set_dps_uds: Set DPS Unique Device Secret (UDS) for X.509 certificates..
 - set_az_iotdps: Set DPS Symmetric Key. Format: "DPSEndpoint=global.azure-devices-provisioning.net;IdScope=XXX;DeviceId=XXX;SymmetricKey=XXX".
 - enable_secure: Enable secure channel between AZ3166 and secure chip.

# set_az_iothub HostName=xxx.azure-devices.net;DeviceId=xxx;SharedAccessKey=xxx
INFO: Set Azure Iot hub connection string successfully.

#
************************************************
** MXChip - Microsoft IoT Developer Kit **
************************************************
You can 1. press Button A and reset to enter configuration mode.
        2. press Button B and reset to enter AP mode.

Wi-Fi xxx connected.
Now is (UTC): Mon Feb 24 03:02:21 2020

2020-02-24 03:02:23 INFO:  IoThub Version: 1.3.6

2020-02-24 03:02:30 INFO:  >>>Connection status: connected
2020-02-24 03:02:35 INFO:  >>>IoTHubClient_LL_SendEventAsync accepted message for transmission to IoT Hub.
2020-02-24 03:02:36 INFO:  >>>Confirmation[0] received for message tracking id = 0 with result = IOTHUB_CLIENT_CONFIRMATION_OK
2020-02-24 03:02:41 INFO:  >>>IoTHubClient_LL_SendEventAsync accepted message for transmission to IoT Hub.
2020-02-24 03:02:42 INFO:  >>>Confirmation[1] received for message tracking id = 1 with result = IOTHUB_CLIENT_CONFIRMATION_OK
`
This command do not need these quotes with connection string.

Secondly, you can use command in vscode to configue it, the link is here https://docs.microsoft.com/en-us/samples/azure-samples/mxchip-iot-devkit-get-started/sample/

And if someone have the same issue, we can talk together in this gitter https://gitter.im/Microsoft/azure-iot-developer-kit

JerryYangKai, 2020-03-03:
Close issue without reply for seven days

closed 2020-03-03 by JerryYangKai

andrewwhitten, 2020-12-27:
I am getting exactly the same issue 9 months later. "Failed to send telemetry data: 400" - my connection string is correct, and I have tried with both 1.6.5 and 2.0.0 firmware