microsoft/devkit-sdk #1085: Unable to Upload Device Code due to missing files within openocd dependency

tyh77, 2021-07-07:
Received following error while trying to build + upload solution

> Loading configuration...
> Initializing packages...
> Preparing boards...
> Verifying...
> /Users/<user>/Library/Arduino15/packages/AZ3166/tools/arm-none-eabi-gcc/5_4-2016q3/bin/../lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/bin/ld: warning: changing start of section .data by 4 bytes
> /Users/<user>/Library/Arduino15/packages/AZ3166/tools/arm-none-eabi-gcc/5_4-2016q3/bin/../lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/bin/ld: warning: changing start of section .data by 4 bytes
> /Users/<user>/Library/Arduino15/packages/AZ3166/tools/arm-none-eabi-gcc/5_4-2016q3/bin/../lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/bin/ld: warning: changing start of section .data by 4 bytes
> Uploading...
> java.io.IOException: Cannot run program "/Users/<user>/Library/Arduino15/packages/AZ3166/tools/openocd/0.10.0/macosx/bin/openocd": error=2, No such file or directory
> 	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1048)
> 	at java.lang.Runtime.exec(Runtime.java:621)
> 	at java.lang.Runtime.exec(Runtime.java:486)
> 	at processing.app.helpers.ProcessUtils.exec(ProcessUtils.java:11)
> 	at cc.arduino.packages.Uploader.executeUploadCommand(Uploader.java:129)
> 	at cc.arduino.packages.uploaders.SerialUploader.runCommand(SerialUploader.java:383)
> 	at cc.arduino.packages.uploaders.SerialUploader.uploadUsingPreferences(SerialUploader.java:197)
> 	at cc.arduino.UploaderUtils.upload(UploaderUtils.java:77)
> 	at processing.app.Base.<init>(Base.java:439)
> 	at processing.app.Base.main(Base.java:150)
> Caused by: java.io.IOException: error=2, No such file or directory
> 	at java.lang.UNIXProcess.forkAndExec(Native Method)
> 	at java.lang.UNIXProcess.<init>(UNIXProcess.java:247)
> 	at java.lang.ProcessImpl.start(ProcessImpl.java:134)
> 	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1029)
> 	... 9 more
> An error occurred while uploading the sketch
> [Warning] Failed to generate IntelliSense configuration.
> [Error] Uploading sketch 'RemoteMonitoringV2.ino': Exit with code=1

It appears that the **openocd** dependency doesn't seem to be properly prepared? It's expecting a macosx directory but the path only contains the following:

- AUTHORS
- BUGS
- COPYING
- ChangeLog
- Doxyfile.in
- HACKING
- INSTALL
- Makefile.am
- Makefile.in
- NEWS
- NEWS-0.2.0
- NEWS-0.3.0
- NEWS-0.4.0
- NEWS-0.5.0
- NEWS-0.6.0
- NEWS-0.7.0
- NEWS-0.8.0
- NEWS-0.9.0
- NEWTAPS
- README
- README.OSX
- README.Windows
- TODO
- aclocal.m4
- compile
- config.guess
- config.h.in
- config.sub
- config_subdir.m4
- configure
- configure.ac
- contrib

depcomp
doc
install-sh
jimtcl
ltmain.sh
mdate-sh
missing
src
tcl
texinfo.tex
tools

tyh77, 2021-07-12:
@hellyzh @JerryYangKai was wondering if any of you would be able to take a quick look at the above issue? I believe my situation is not unique and would love to get some help on this 

It would also help others who have faced similar issues before https://github.com/microsoft/azure-iot-developer-kit/issues/524

erikmetzinfo, 2021-08-08:
Same issue here

pmarshDius, 2021-09-10:
Has anybody go a resolution for this yet.  I'm seeing the same issue

xalikoutis, 2022-03-12:
same here any solution?